Flamingo-tech/Open-AIR #4: Support for bypass

decramy, 2023-05-08:
My ventilationbox with Heat-Exchange (Dutch: WTW) supports bypassing the heat exchange for hot days (so the heat from the house is not put back into the house).
Also there is a pre-heating option available for this device (I dont have that).

Can you add one or more relais (with according software options) for this? Maybe with configurable names?

Flamingo-tech, 2025-07-06:
Hey @decramy for now WTWs are not supported! All supported models are listed on the open AIR Mini page : https://github.com/Flamingo-tech/Open-AIR/tree/main/Open%20AIR%20Mini

closed 2025-07-06 by Flamingo-tech

decramy, 2025-07-06:
erm yes... Thats why I mentioned it ;-) Not a bug report, more a feature request...