Flamingo-tech/Open-AIR #7: Improvement for Open Air Valve YAML

roderik-maker, 2024-02-27:
I think the YAML file for the open air valve could benefit from adding some more checks. I made some additions and added a load of comments to the file. Wasn't sure if I should post here, or on the open-air-valve page, opted for the open-air-valve page but here is a [link to the thread](https://github.com/thomasvnl/esphome-open-air-valve/issues/3#issue-2156998379).

peterfortuin, 2025-02-19:
@roderik-maker, it seems that your link is not working any more.

roderik-maker, 2025-07-07:
I'll try find some time tonight to share it again

roderik-maker, 2025-07-09:
Its been a very bussy week and I can't find the original either anymore. Here is my current config for 1 of my valves, I think is the best version I currently have. I'll try find time on Saturday to clean it up and add some comments and explanations.

```
substitutions:
  device_friendly_name: "Valve-1"
  valve_as_number_id: "valve_capacity"
  my_stepper_id: "my_stepper"
  valve_capacity_lookup_table: id(valve_capacity_lookup_table)

esphome:
  name: open-air-valve-1
  friendly_name: ${device_friendly_name}
  name_add_mac_suffix: False
  
  on_boot:
    - priority: 700
      then:
      - stepper.set_acceleration:
          id: my_stepper
          acceleration: inf
      - stepper.set_deceleration:
          id: my_stepper
          deceleration: inf
      - stepper.report_position:
          id: my_stepper
          position: 0
      - stepper.set_target:
          id: my_stepper
          target: !lambda 'return id(valve_capacity_lookup_table)[10];'
      - wait_until:
          condition:
            lambda:
              return id(my_stepper).current_position == id(my_stepper).target_position;
      - stepper.set_speed:
          id: my_stepper
          speed: 150 steps/s
      - if:
          condition:
            binary_sensor.is_off: valve_homing_switch
          then:
          - while:
              condition:
                binary_sensor.is_off: valve_homing_switch
              then:
                - stepper.report_position:
                    id: my_stepper
                    position: 0
                - stepper.set_target:
                    id: my_stepper
                    target: -1
                - wait_until:
                    condition:
                      lambda:
                        return id(my_stepper).current_position <= id(my_stepper).target_position;
      - stepper.report_position:
          id: my_stepper
          position: 0
      - stepper.set_target:
          id: my_stepper
          target: 0
      - stepper.set_speed:
          id: my_stepper
          speed: 350 steps/s

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
[... 161 lines cut ...]
      - wait_until:
          condition:
            lambda:
              return id(my_stepper).current_position == id(my_stepper).target_position;
      - if:
          condition:
            binary_sensor.is_off: valve_homing_switch
          then:
          - while:
              condition:
                binary_sensor.is_off: valve_homing_switch
              then:
                - stepper.report_position:
                    id: my_stepper
                    position: 0
                - stepper.set_target:
                    id: my_stepper
                    target: -1
                - wait_until:
                    condition:
                      lambda:
                        return id(my_stepper).current_position <= id(my_stepper).target_position;

```
The problem is, if it is really stuck, it'll stay stuck. I still need to add a loop counter that throws an error after x-step attempts and then blocks the stepper motor from moving to prevent damage.