Flamingo-tech/Open-AIR #18: ESPhome update 2025.2 gives an error

sanderlv, 2025-02-19:
```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/esp32-duco-01.yaml...
INFO Detected timezone 'Europe/Amsterdam'
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

sensor.custom: [source /config/esphome/esp32-duco-01.yaml:106]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: |-
    auto sht20 = new SHT20();
    App.register_component(sht20);
    return {sht20->temperature_sensor, sht20->humidity_sensor, sht20->vpd_sensor, sht20->dew_point_sensor};
  sensors: 
    - name: T
      internal: True
      id: sensor_temperature
      accuracy_decimals: 2
    - name: H
```
Is this something "easy" to change?

https://esphome.io/guides/contributing#a-note-about-custom-components

thomasvnl, 2025-02-19:
Hi Sander,

The easiest for now would be to remain on a lower version of ESPHome for
this device. I do not have a better answer at this time.

peterfortuin, 2025-02-19:
I'm wondering what kind of config yaml you have, because the current yaml's in the repo don't have custom components.

thomasvnl, 2025-02-21:
@sanderlv I see you've reacted with an unhappy frowning smiley, but my suggestion to lower the version of ESPHome (for this specific device) would be the "Is this something "easy" to change?" solution (for now).

Flamingo-tech, 2025-02-21:
Please use this integration method for the SHT-20 Sensor: [Click](https://github.com/Flamingo-tech/Open-AIR/tree/main/Open%20AIR%20Mini/Software#sensor-support-SHT-20)

Example:

`external_components:
  - source: github://dmaasland/esphome@sht2x
    components: [ sht2x ]`

`
  - platform: sht2x
    i2c_id: i2c_sensor_1
    temperature:
      name: "Open AIR Mini x Temperature"
      id: air_temperature
      accuracy_decimals: 2
    humidity:
      name: "Open AIR Mini x Humidity"
      id: air_humidity
      accuracy_decimals: 2
    update_interval: 30s
`

This should solve your issue. The method you are trying to use was the old method. If you need help with your new YAML please mail : flamingo-tech@protonmail.com

closed 2025-02-21 by Flamingo-tech

sanderlv, 2025-03-01:
Have a look here: this solves the issue temporarily:

https://github.com/robertklep/esphome-custom-component